danieleborgo/OctoPrint-GPIOStatus #1: Feature requests

RayPS, 2022-01-10:
Thanks for making such an amazing plugin for OctoPrint!
It's super useful for advanced users like us and I enjoy it!
However, I had some feature request for this plugin.

1. Accessible as a panel like other plugins do. GPIO Status is used very frequently, it's difficult to check through settings window every time.
2. Ability to describe/name/define a pin. So we can remember what's attached to the pin.
3. An option to toggle between BOARD/BCM pin numbering. (Physical Pin)
4. An option to hide power pins in list view. (3V3/5V/GND) or we can call it "Show GPIO pins only".
[image]

InoSiX, 2022-01-11:
I would like to double up on the thanks here! Great idea and interesting it took so long for someone to recognize a use case for this!

I would also like to suggest all 4 of the items mentioned here. These would be great additions.

danieleborgo, 2022-01-11:
Thank you both for your appreciation.
Yesterday I implemented points 1, 3, and 4, I published these few minutes ago.
I'm definitely going to start working on feature 2, it is a very interesting idea.

[image]

InoSiX, 2022-01-11:
> 3\. option to toggle between BOARD/BCM pin numbering. (Physical Pin)

Dope dude! Great turn around time. I imagine this plugin being adopted rapidly. Thanks for your work!

danieleborgo, 2022-01-12:
I implemented also feature 2. I just uploaded the code.

InoSiX, 2022-01-13:
Work like yours makes Octoprint singlehandedly the best upgrade for a 3D printer. Seriously. Thanks.

closed 2022-01-13 by danieleborgo