danieleborgo/OctoPrint-GPIOStatus #2: Not working on RPI4

biancapascal, 2022-01-10:
Is this tested on a older RPI? 
On the RPI4 it dos'nt work, get the message **Command raspi-gpio not found. Please install this on to Raspberry**
Maybe a install text in main would help if it should work on a RPI4.

danieleborgo, 2022-01-11:
Do you mean that the plugin needs to specify the command to execute to install raspi-gpio?
Thank you for your feedback

biancapascal, 2022-01-11:
So far as i know everything what is mentiond what is needed to run that plugin is installed, but when i open the plugin it says **Command raspi-gpio not found. Please install this on to Raspberry**
So there for my question on witch raspberry pi is it tested. On the Raspberry Pi it dos'nt work, tested on 2 different pi's and printers.

Braunc55, 2022-01-12:
Works fine on my RPI 4 Rev b03111. Let me know if there is anything I can assist to help troubleshoot.

biancapascal, 2022-01-12:
how did you install everything? on my image of octoprint what i downloaden it dos not work, even updated everything

Braunc55, 2022-01-12:
> how did you install everything? on my image of octoprint what i downloaden it dos not work, even updated everything

I installed the GPIOStatus plugin via the Plugin Manager. I am running OctoPrint 1.7.2 any python 3.7.3.

danieleborgo, 2022-01-12:
GPIO Status uses the command _raspi-gpio_ and usually it is preinstalled on Raspberry images, similarly to _raspi-config_. In your case you don't have it, so you can install it logging in your Raspberry via a terminal, like via SSH for instance, and install it via _apt_.

Remember that the plugin needs the command _raspi-gpio_ on its terminal and **not** the _npm_ package that has the same name.

biancapascal, 2022-01-12:
sudo apt-get install raspi-gpio

closed 2022-01-12 by danieleborgo