danieleborgo/OctoPrint-GPIOStatus #3: With this plugin enabled, GPIO control stops working

omega015, 2022-01-12:
With this plugin enabled, GPIO control and PSU control stops working (no psu control button and doesnt change gpio state). The moment I disable GPIOStatus those two work again. When enabled the power icon / lightening bolt no longer shows and a big emergency stop button appears (along with the one that was already there) So it seems to affect that too (at least visually).

OctoPrint 1.7.2
OctoPi 0.17.0
PSU Control (1.0.6) (with PSU Control - RPi.GPIO (1.0.4))
GPIO Control (1.0.8)
Simple Emergency Stop (1.0.5)

All up to date.

danieleborgo, 2022-01-12:
We are using the same OctoPrint version but I need to test more in details GPIO Status with these ones.

danieleborgo, 2022-01-14:
Have you check if there are any error in the OctoPrint logger files?

omega015, 2022-01-14:
No warnings or errors only generic info logs about everything starting up etc...

InoSiX, 2022-01-17:
Hiya,
For what it is worth, the digital (open/close) contacts I have on my RPI4 still work with this installed.

omega015, 2022-01-17:
@InoSiX Are you using any of the other plugins at the same time? specifically PSU Control (1.0.6) (with PSU Control - RPi.GPIO (1.0.4)) or GPIO Control (1.0.8)?

InoSiX, 2022-01-18:
PSU Control (1.0.6) + PSU Control - GPIO (1.0.4), indeed, yes. I'm currently using it to turn on/off a solid state (3v) relay to power the printer. I can confirm it works on PIN 23, at the very least.

omega015, 2022-01-28:
I cannot comment on InoSix but for my problem I am setup to control GPIO 17 / Pin 11 as my power switch.

danieleborgo, 2022-02-10:
@omega015 Could you try to uninstall this plugin and redo the installation with the newest version?

omega015, 2022-02-10:
@danieleborgo Looks like the update has resolved this issue, thank you. I notice that all status are stuck on updating though, should I log this as a separate issue?

danieleborgo, 2022-02-11:
> @danieleborgo Looks like the update has resolved this issue, thank you. I notice that all status are stuck on updating though, should I log this as a separate issue?

Is your problem similar to [this issue](https://github.com/danieleborgo/OctoPrint-GPIOStatus/issues/4)?

danieleborgo, 2022-02-18:
@omega015 If your problem is similar to that issue, can I close this one?
Thank you

omega015, 2022-02-18:
Hi @danieleborgo sorry for the delay I am traveling and haven't had chance to compare but yes please close this one and I'll look at the other issue and see if that relates to mine or not.

Thank you for your work.

closed 2022-02-20 by danieleborgo