danieleborgo/OctoPrint-GPIOStatus #4: Everything says "Updating..."

G-Ray2001, 2022-02-07:
This plug in isn't showing anything on my system. Everything shows "Updating...", and that's it. I have uninstalled and reinstalled the plug in but no change. I do have GPIO enabled, as I use one of the pins for filament detection, and get the GPIO status on a SSH window.

Any suggestions?
Thanks

danieleborgo, 2022-02-10:
Do you have errors or exceptions in the browser developer console? Or maybe anything in OctoPrint logs?

G-Ray2001, 2022-02-11:
Hello,
No errors shown in Octoprint, but the octoprint.log states that "No such file or directory: 'raspi-gpio': 'raspi-gpio'".

Is this supposed to be created when installing the GPIO-Status plugin? I know the raspi-config console, but not familiar with "raspi-gpio".

Thanks

danieleborgo, 2022-02-11:
Your problem is the fact you have to install _raspi-gpio_ on your Raspberry.
Anyway, the plugin should tell you this but it doesn't because of a bug I already identified. I'm going to solve this as soon as possible.

danieleborgo, 2022-02-11:
I solved the bug, now it newly notifies the user:
[image]

G-Ray2001, 2022-02-12:
Hello,
Okay, I finally figured out how to install raspi-gpio, I found one place on
Github and had to download, compile,and install it manually. Ran commands
from SSH:

raspi-gpio get:  I see pin information

raspi-gpio funcs: I see pin information

raspi-gpio raw: I see what appears to be hexadecimal information

Now, when I access the GPIO-Status tab in Octoprint, I click "Refresh" and
get "Failed to Retrieve" on them.

Looking at the Octoprint.log file now, I see this when I refreshed:

2022-02-11 22:44:28,920 - octoprint.plugins.gpiostatus - INFO - Refresh
request received
2022-02-11 22:44:28,950 - octoprint.server.api - ERROR - Error while
executing SimpleApiPlugin gpiostatus
Traceback (most recent call last):
  File
"/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/__init__.py",
line 158, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File
"/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py",
line 1737, in wrapper
    return f(*args, **kwargs)
  File
"/home/pi/oprint/lib/python3.7/site-packages/octoprint_gpiostatus/__init__.py",
line 64, in on_api_command
    funcs_required=("funcs" not in data) or data["funcs"]
  File
"/home/pi/oprint/lib/python3.7/site-packages/octoprint_gpiostatus/__init__.py",
line 81, in __get_status
    self.__inject_bcm_data(status["pins"], funcs_required)
  File
"/home/pi/oprint/lib/python3.7/site-packages/octoprint_gpiostatus/__init__.py",
line 155, in __inject_bcm_data
    bcm_pins_status = self.__get_bcm_pins_status(funcs_required)
  File
"/home/pi/oprint/lib/python3.7/site-packages/octoprint_gpiostatus/__init__.py",
line 169, in __get_bcm_pins_status
    bcm_pins_status = self.__get_raw_status()
  File
"/home/pi/oprint/lib/python3.7/site-packages/octoprint_gpiostatus/__init__.py",
line 203, in __get_raw_status
    ).split("\n")
  File
"/home/pi/oprint/lib/python3.7/site-packages/octoprint_gpiostatus/__init__.py",
line 199, in <listcomp>
    for config in [
IndexError: list index out of range

Thanks,
Ray

danieleborgo, 2022-02-12:
If you uninstall the plugin, reboot and install again, does it works?

G-Ray2001, 2022-02-12:
No, same response as before

danieleborgo, 2022-02-13:
It seems the plugin is unable to successfully execute the command _raspi-gpio get 0-27_, which is strange since you have installed it.
Could you give me more information about your Raspberry, your OS and other things that may be useful?
Thank you

G-Ray2001, 2022-02-13:
I'm not sure what is useful for you, but okay, it's a Raspberry pi 3B+ and
it's running:

OctoPi 0.17.0   OctoPrint 1.7.3    Python 3.7.3
All plug-ins installed are up-to-date.

A 32GB high speed micro SD card

I attached a screenshot of the SSH running that very same command you wrote
me.

Thank you.

   *- OctoPrint 1.7.3 - Python 3.7.3 - OctoPi 0.17.0*

danieleborgo, 2022-02-13:
> I'm not sure what is useful for you, but okay, it's a Raspberry pi 3B+ and it's running: OctoPi 0.17.0 OctoPrint 1.7.3 Python 3.7.3 All plug-ins installed are up-to-date. A 32GB high speed micro SD [...]

I cannon see the screenshot 🤔

G-Ray2001, 2022-02-13:
Well, it's just confirmation that the command raspi-gpio get 0-27 works in
the command shell.

Anyway, I tested the GPIO-Status plug in on a clean Octoprint image, newest
of everything, (OctoPi 0.18.0, Python 3.7.3, Octoprint 1.7.3, with no other
plugins installed, and it worked just fine.

danieleborgo, 2022-02-18:
Does this plugin write _Plugin ready_ after initialization?

danieleborgo, 2022-05-28:
Hi,
if you fixed the problem, can I close this issue? Or do you still have problems?

closed 2022-05-30 by danieleborgo

mholzma, 2025-07-29:
I get the same error for what it is worth.