danieleborgo/OctoPrint-GPIOStatus #6: Can't edit notes

EricBenevent, 2022-08-17:
Thanks a lot for creating this very usefull plugin !
I can't figure out how to edit the notes to name the pins according to my setup, what am I doing wrong ?
GPIO Status 0.2.1
Octoprint Version 1.8.2
Rasberry PI 3B

EricBenevent, 2022-08-21:
Edit : 
It does not work when using Firefox 103.0.2 (64 bits)
It does work fine when using Chrome Version 104.0.5112.82 (Build officiel) (64 bits)
On a windows 10 profesional version 21H2
So I will just use Chrome for editing for now !

danieleborgo, 2022-08-23:
Hi @EricBenevent,
thank you for the feedback. I just uploaded a new fix to solve this problem, is it now working also for you?
I tryed different browsers and I noticed that this problem was only on FireFox.

EricBenevent, 2022-08-25:
Hello @danieleborgo, yes, it does work as well on Firefox now, thanks for maintaining this nice tool !
I installed an Arduino dedicated for safety of the printer, and I use your pluggin for diagnostic via the GPIO inputs, so I can easilly figure out what trigerred the Arduino to kill power to the printer.

closed 2022-08-25 by EricBenevent