danieleborgo/OctoPrint-GPIOStatus #7: Feature Request: RockPI / Generic GPIO Status

sammcj, 2023-02-08:
Great plugin! It would be awesome if it worked to retrieve generic GPIO statuses, this would add the ability for people with RockPi / BananaPi etc... to use this.

The standard `gpiod` package provides tooling such as gpiodetect and gpioinfo.

Example output from my Rock Pi 4 SE:

```
rockpi:~:# gpiodetect
gpiochip0 [gpio0] (32 lines)
gpiochip1 [gpio1] (32 lines)
gpiochip2 [gpio2] (32 lines)
gpiochip3 [gpio3] (32 lines)
gpiochip4 [gpio4] (32 lines)

rockpi:~:# gpioinfo
gpiochip0 - 32 lines:
	line   0:      unnamed       unused   input  active-high
	line   1:      unnamed       unused   input  active-high
	line   2:      unnamed       unused   input  active-high
	line   3:      unnamed       unused   input  active-high
	line   4:      unnamed "host-wakeup" input active-high [used]
	line   5:      unnamed       unused   input  active-high
	line   6:      unnamed       unused   input  active-high
	line   7:      unnamed         "cd"   input   active-low [used]
	line   8:      unnamed       unused   input  active-high
	line   9:      unnamed   "shutdown"  output  active-high [used]
	line  10:      unnamed      "reset"  output   active-low [used]
	line  11:      unnamed       unused   input  active-high
	line  12:      unnamed       unused   input  active-high
	line  13:      unnamed       unused   input  active-high
	line  14:      unnamed       unused   input  active-high
	line  15:      unnamed       unused   input  active-high
	line  16:      unnamed       unused   input  active-high
	line  17:      unnamed       unused   input  active-high
	line  18:      unnamed       unused   input  active-high
	line  19:      unnamed       unused   input  active-high
	line  20:      unnamed       unused   input  active-high
	line  21:      unnamed       unused   input  active-high
	line  22:      unnamed       unused   input  active-high
	line  23:      unnamed       unused   input  active-high
	line  24:      unnamed       unused   input  active-high
	line  25:      unnamed       unused   input  active-high
	line  26:      unnamed       unused   input  active-high
	line  27:      unnamed       unused   input  active-high
	line  28:      unnamed       unused   input  active-high
	line  29:      unnamed       unused   input  active-high
	line  30:      unnamed       unused   input  active-high
	line  31:      unnamed       unused   input  active-high
gpiochip1 - 32 lines:
	line   0:      unnamed       unused   input  active-high
	line   1:      unnamed       unused   input  active-high
	line   2:      unnamed       unused   input  active-high
	line   3:      unnamed "vcc5v0-typec-regulator" output active-high [used]
	line   4:      unnamed       unused   input  active-high
	line   5:      unnamed       unused   input  active-high
	line   6:      unnamed       unused   input  active-high
	line   7:      unnamed       unused   input  active-high
	line   8:      unnamed       unused   input  active-high
	line   9:      unnamed       unused   input  active-high
	line  10:      unnamed       unused   input  active-high
	line  11:      unnamed       unused   input  active-high
	line  12:      unnamed       unused   input  active-high
	line  13:      unnamed       unused   input  active-high
	line  14:      unnamed       unused   input  active-high
	line  15:      unnamed       unused   input  active-high
	line  16:      unnamed       unused   input  active-high
	line  17:      unnamed       unused   input  active-high
	line  18:      unnamed       unused   input  active-high
	line  19:      unnamed       unused   input  active-high
	line  20:      unnamed       unused   input  active-high
[... 105 lines cut ...]
	line  27:      unnamed         "ep"  output  active-high [used]
	line  28:      unnamed       unused   input  active-high
	line  29:      unnamed       unused   input  active-high
	line  30:      unnamed       unused   input  active-high
	line  31:      unnamed       unused   input  active-high
```

```dmesg
rockpi:~:# dmesg|grep -i gpio
[    0.125714] rockchip-gpio ff720000.gpio0: probed /pinctrl/gpio0@ff720000
[    0.126527] rockchip-gpio ff730000.gpio1: probed /pinctrl/gpio1@ff730000
[    0.127302] rockchip-gpio ff780000.gpio2: probed /pinctrl/gpio2@ff780000
[    0.128065] rockchip-gpio ff788000.gpio3: probed /pinctrl/gpio3@ff788000
[    0.128866] rockchip-gpio ff790000.gpio4: probed /pinctrl/gpio4@ff790000
[    3.335694] rk808-regulator rk808-regulator: there is no dvs0 gpio
[    3.335730] rk808-regulator rk808-regulator: there is no dvs1 gpio
[    3.368751] dwmmc_rockchip fe320000.mmc: Got CD GPIO

rockpi:~:# ls -la /dev/gpiochip*
crw------- 1 root root 254, 0 Feb  9 08:13 /dev/gpiochip0
crw------- 1 root root 254, 1 Feb  9 08:13 /dev/gpiochip1
crw------- 1 root root 254, 2 Feb  9 08:13 /dev/gpiochip2
crw------- 1 root root 254, 3 Feb  9 08:13 /dev/gpiochip3
crw------- 1 root root 254, 4 Feb  9 08:13 /dev/gpiochip4
```

danieleborgo, 2023-02-10:
Hi,
thank you for the appreciation.

This plugin doesn't use a command to retrieve the data but a Python module. In your Rock PI all the four screens don't work or just some parts?

sammcj, 2023-02-14:
Ah ok, Yeah none of the screens seem to work - it's the same as if I was running it on a machine without GPIO.

No pressure to add it though, I can make do without, I don't want to waste your time.

wesalmon, 2023-07-19:
I would like to second that request.  What is the module you mention in the previous post?  I have several boards to test but I have no skills to create the script.
I have Rock64, Orange Pi zero, le potato.  I am willing to test!
Thank you!

danieleborgo, 2023-08-04:
Thank you @wesalmon for your help.
It's been a while since I worked on this plugin, the problem I need to address now is to find out a way to recover all the data I need, since the commands (_raspi-gpio_ and _raspi-config_) and the Python module (_gpiozero_) don't work on these boards. The command @sammcj has kindly posted is useful to get the pin function (IN or OUT) and their state but this represent sadly just half the data needed.

Another important thing I need is a command or a Python module to recover services status. I'm going to search something in the next weeks but any ideas are welcome since I cannot directly test 😂.