danieleborgo/OctoPrint-GPIOStatus #8: OctoPrint-GPIOStatus plugin does not install!

mc-ott, 2024-02-08:
I tried a few times but the Plugin management log says "Install Unknown: " and the plug-in does not install. @danieleborgo, I think it may be the same problem as in FilamentBuddy plugin last week... 

Goes through the process of installation below:

Installing plugin "GPIO Status" from https://github.com/danieleborgo/OctoPrint-GPIOStatus/archive/master.zip...
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpyjy135ow/OctoPrint-GPIOStatus-master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpyjy135ow/OctoPrint-GPIOStatus-master.zip
Requirement already satisfied: OctoPrint>=1.7.2 in /home/pi/oprint/lib/python3.7/site-packages (from GPIO-Status==0.2.1) (1.9.3)
Requirement already satisfied: gpiozero>=1.6.2 in /home/pi/oprint/lib/python3.7/site-packages (from GPIO-Status==0.2.1) (2.0)
Requirement already satisfied: RPi.GPIO>=0.7.0 in /home/pi/oprint/lib/python3.7/site-packages (from GPIO-Status==0.2.1) (0.7.1)
Requirement already satisfied: flask>=1.1.4 in /home/pi/oprint/lib/python3.7/site-packages (from GPIO-Status==0.2.1) (2.2.5)
Requirement already satisfied: Jinja2>=3.0 in /home/pi/oprint/lib/python3.7/site-packages (from flask>=1.1.4->GPIO-Status==0.2.1) (3.1.2)
Requirement already satisfied: Werkzeug>=2.2.2 in /home/pi/oprint/lib/python3.7/site-packages (from flask>=1.1.4->GPIO-Status==0.2.1) (2.2.3)
Requirement already satisfied: itsdangerous>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from flask>=1.1.4->GPIO-Status==0.2.1) (2.1.2)
Requirement already satisfied: importlib-metadata>=3.6.0 in /home/pi/oprint/lib/python3.7/site-packages (from flask>=1.1.4->GPIO-Status==0.2.1) (6.7.0)
Requirement already satisfied: click>=8.0 in /home/pi/oprint/lib/python3.7/site-packages (from flask>=1.1.4->GPIO-Status==0.2.1) (8.1.7)
Requirement already satisfied: colorzero in /home/pi/oprint/lib/python3.7/site-packages (from gpiozero>=1.6.2->GPIO-Status==0.2.1) (2.0)
Requirement already satisfied: typing-extensions>=3.6.4 in /home/pi/oprint/lib/python3.7/site-packages (from importlib-metadata>=3.6.0->flask>=1.1.4->GPIO-Status==0.2.1) (4.7.1)
Requirement already satisfied: zipp>=0.5 in /home/pi/oprint/lib/python3.7/site-packages (from importlib-metadata>=3.6.0->flask>=1.1.4->GPIO-Status==0.2.1) (3.15.0)
Requirement already satisfied: MarkupSafe>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from Jinja2>=3.0->flask>=1.1.4->GPIO-Status==0.2.1) (2.1.3)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (2.0)
Requirement already satisfied: OctoPrint-PiSupport>=2023.5.24 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (2023.10.10)
Requirement already satisfied: pathvalidate<3,>=2.5.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (2.5.2)
Requirement already satisfied: wheel in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (0.36.2)
Requirement already satisfied: PyYAML<7,>=6.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (6.0.1)
Requirement already satisfied: Babel<2.13,>=2.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (2.12.1)
Requirement already satisfied: sarge==0.1.7.post1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (0.1.7.post1)
Requirement already satisfied: passlib<2,>=1.7.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (1.7.4)
Requirement already satisfied: psutil<6,>=5.9.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (5.9.5)
Requirement already satisfied: regex in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (2020.11.13)
Requirement already satisfied: tornado<7,>=6.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (6.2)
Requirement already satisfied: unidecode in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (0.4.21)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (2021.10.11)
Requirement already satisfied: cachelib<0.11,>=0.10.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (0.10.2)
Requirement already satisfied: pkginfo<2,>=1.9.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (1.9.6)
Requirement already satisfied: feedparser<7,>=6.0.10 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (6.0.10)
Requirement already satisfied: watchdog<3,>=2.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (2.3.1)
Requirement already satisfied: argon2-cffi<22,>=21.3.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (21.3.0)
Requirement already satisfied: zeroconf==0.39.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (0.39.4)
Requirement already satisfied: frozendict<3,>=2.3.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (2.3.8)
Requirement already satisfied: filetype<2,>=1.2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (1.2.0)
Requirement already satisfied: colorlog<7,>=6.7.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (6.7.0)
Requirement already satisfied: markdown<4,>=3.4.3 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (3.4.4)
Requirement already satisfied: pydantic==1.10.12 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (1.10.12)
Requirement already satisfied: websocket-client<2,>=1.5.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (1.6.1)
Requirement already satisfied: pyserial<4,>=3.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (3.5)
Requirement already satisfied: blinker<2,>=1.6.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (1.6.3)
Requirement already satisfied: Flask-Limiter<4,>=3.3.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (3.5.0)
Requirement already satisfied: pylru<2,>=1.2.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (1.2.1)
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (2021.2.23)
Requirement already satisfied: zipstream-ng<2.0.0,>=1.5.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (1.7.1)
Requirement already satisfied: netifaces<1,>=0.11 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (0.11.0)
Requirement already satisfied: netaddr<0.9,>=0.8 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (0.8.0)
Requirement already satisfied: requests<3,>=2.28.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (2.31.0)
Requirement already satisfied: sentry-sdk<2,>=1.19.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (1.31.0)
Requirement already satisfied: wrapt<1.16,>=1.15 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (1.15.0)
Requirement already satisfied: Flask-Babel<4,>=3.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (3.1.0)
Requirement already satisfied: class-doc<0.3,>=0.2.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (0.2.6)
Requirement already satisfied: emoji<3,>=2.2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>=1.7.2->GPIO-Status==0.2.1) (2.8.0)
[... cut ...]
Created wheel for GPIO-Status: filename=GPIO_Status-0.2.1-py3-none-any.whl size=370746 sha256=b319f00d4a156aee009587516ac042a20a7fe6df4b325538b4d37b14ef46817c
Stored in directory: /tmp/pip-ephem-wheel-cache-7wykl5ob/wheels/58/5a/fb/a89d1a627496fd4e2d184b342a4554ae23bba4d71551391c15
Successfully built GPIO-Status
Installing collected packages: GPIO-Status
Attempting uninstall: GPIO-Status
Found existing installation: GPIO-Status 0.2.1
Uninstalling GPIO-Status-0.2.1:
Successfully uninstalled GPIO-Status-0.2.1
Successfully installed GPIO-Status-0.2.1
Done!

I am still running OctoPrint on Raspberry Pi 4B (OctoPrint 1.9.3 Python 3.7.3 OctoPi 0.18).

danieleborgo, 2024-02-09:
Yes, the problem is the same but here the plugin is way more dependend to _gpiozero_ than in FilamentBuddy, so the fix I temporarily applied there won't work here. The solution that always works in this case is to update Python or OctoPi.

Nevertheless, I checked on _gpiozero_ repository and they're aware of this problem, so I think it is better to wait for their response, if you cannot update your system.

mc-ott, 2024-02-09:
Ok, thanks, I will update the Octopi to 1.0.0 this weekend. Michel —————————

danieleborgo, 2024-02-09:
Cool, this should solve all the problems with _gpiozero._
Let me know then if both GPIO Status and FilamentBuddy work properly or if they logs some errors.

mc-ott, 2024-02-09:
👍🏻—————————

danieleborgo, 2024-02-13:
Do the plugins work properly?

mc-ott, 2024-02-15:
Yes, they install with no errors and work OK since I installed OctoPrint 1.9.3, Python 3.9.2, OctoPi* 1.0.0cam (build 2023.10.09.154319).

closed 2024-02-15 by mc-ott