danieleborgo/OctoPrint-GPIOStatus #9: GPIO Status seems not working with Raspberry Compute Module 5

NilsRo, 2025-03-17:
Hi,

I got the following error after restoring my OctoPrint to a CM5. Nothing related to the GPIO seems working fine. Can see the following error.

Cheers,
Nils

```
2025-03-17 23:11:37,795 - octoprint.plugins.SlicerEstimator - DEBUG - Payload: {'username': 'octopi'}
2025-03-17 23:11:37,844 - octoprint.plugins.softwareupdate - INFO - Fetching check overlays from https://plugins.octoprint.org/update_check_overlay.json
2025-03-17 23:11:39,125 - octoprint.server.util.sockjs - INFO - User octopi logged in on the socket from client 192.168.177.109
2025-03-17 23:11:39,134 - octoprint.plugins.SlicerEstimator - DEBUG - Event received: ClientAuthed
2025-03-17 23:11:39,134 - octoprint.plugins.SlicerEstimator - DEBUG - Payload: {'username': 'octopi', 'remoteAddress': '192.168.177.109'}
2025-03-17 23:11:39,434 - octoprint.plugins.announcements - INFO - Loaded channel _important from https://octoprint.org/feeds/important.xml in 0.23s
2025-03-17 23:11:39,518 - octoprint.plugins.pluginmanager - INFO - Loaded plugin notices data from https://plugins.octoprint.org/notices.json
2025-03-17 23:11:39,621 - octoprint.plugins.gpiostatus - INFO - Refresh request received
2025-03-17 23:11:39,624 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin gpiostatus
Traceback (most recent call last):
  File "/opt/octopi/oprint/lib/python3.11/site-packages/octoprint/server/api/__init__.py", line 161, in pluginCommand
    response = api_plugin.on_api_command(command, data)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/octopi/oprint/lib/python3.11/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^
  File "/opt/octopi/oprint/lib/python3.11/site-packages/octoprint_gpiostatus/__init__.py", line 61, in on_api_command
    return jsonify(self.__get_status(
                   ^^^^^^^^^^^^^^^^^^
  File "/opt/octopi/oprint/lib/python3.11/site-packages/octoprint_gpiostatus/__init__.py", line 81, in __get_status
    self.__inject_bcm_data(status["pins"], funcs_required)
                           ~~~~~~^^^^^^^^
TypeError: 'NoneType' object is not subscriptable
2025-03-17 23:11:39,642 - octoprint.plugins.klipper - INFO - list_cfg_files Path: /home/pi/.octoprint/data/klipper/configs/*
2025-03-17 23:11:39,643 - octoprint.plugins.klipper - INFO - list_cfg_files 1: /home/pi/.octoprint/data/klipper/configs/printer.cfg
2025-03-17 23:11:39,650 - octoprint.plugins.klipper - INFO - list_cfg_files Path: /home/pi/printer_data/config/*.cfg
2025-03-17 23:11:39,651 - octoprint.plugins.klipper - INFO - list_cfg_files 1: /home/pi/printer_data/config/printer.cfg
2025-03-17 23:11:39,657 - octoprint.plugins.klipper - INFO - get_cfg_files Path: /home/pi/printer_data/config/printer.cfg
2025-03-17 23:11:39,755 - octoprint.plugins.announcements - INFO - Loaded channel _releases from https://octoprint.org/feeds/releases.xml in 0.2s
2025-03-17 23:11:39,778 - octoprint.plugins.tracking - INFO - Sent tracking event webui_load, payload: {'browser_name': 'Edge', 'browser_version': '134.0.0.0', 'os_name': 'Windows', 'os_version': '10'}
2025-03-17 23:11:39,972 - octoprint.plugins.announcements - INFO - Loaded channel _blog from https://octoprint.org/feeds/octoblog.xml in 0.19s
2025-03-17 23:11:40,032 - octoprint.util.pip - INFO - Using "/opt/octopi/oprint/bin/python -m pip" as command to invoke pip
2025-03-17 23:11:40,032 - octoprint.util.pip - INFO - pip installs to /opt/octopi/oprint/lib/python3.11/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2025-03-17 23:11:40,032 - octoprint.util.pip - INFO - ==> pip ok -> yes
2025-03-17 23:11:40,234 - octoprint.plugins.SlicerEstimator - DEBUG - Event received: DisplayLayerProgress_timerTrigger
2025-03-17 23:11:40,234 - octoprint.plugins.SlicerEstimator - DEBUG - Payload: {'updateReason': 'timerTrigger', 'totalLayer': '-', 'currentLayer': '-', 'lastLayerDuration': '-', 'lastLayerDurationInSeconds': '-', 'averageLayerDuration': '-', 'averageLayerDurationInSeconds': '-', 'currentHeight': '-', 'currentHeightFormatted': '-', 'totalHeight': '-', 'totalHeightFormatted': '-', 'feedrate': '-', 'feedrateG0': '-', 'feed>
2025-03-17 23:11:40,302 - octoprint.plugins.announcements - INFO - Loaded channel _plugins from https://plugins.octoprint.org/feed.xml in 0.3s
2025-03-17 23:11:40,599 - octoprint.plugins.announcements - INFO - Loaded channel _octopi from https://octoprint.org/feeds/octopi.xml in 0.27s
2025-03-17 23:11:41,894 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2025-03-17 23:12:02,131 - octoprint.plugins.gpiostatus - INFO - Refresh request received
2025-03-17 23:12:02,132 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin gpiostatus
Traceback (most recent call last):
  File "/opt/octopi/oprint/lib/python3.11/site-packages/octoprint/server/api/__init__.py", line 161, in pluginCommand
    response = api_plugin.on_api_command(command, data)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/octopi/oprint/lib/python3.11/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^
  File "/opt/octopi/oprint/lib/python3.11/site-packages/octoprint_gpiostatus/__init__.py", line 61, in on_api_command
    return jsonify(self.__get_status(
                   ^^^^^^^^^^^^^^^^^^
  File "/opt/octopi/oprint/lib/python3.11/site-packages/octoprint_gpiostatus/__init__.py", line 81, in __get_status
    self.__inject_bcm_data(status["pins"], funcs_required)
                           ~~~~~~^^^^^^^^
TypeError: 'NoneType' object is not subscriptable
2025-03-17 23:12:05,673 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from https://plugins.octoprint.org/plugins.json
2025-03-17 23:12:29,835 - octoprint.plugins.pluginmanager - INFO - Loaded plugin notices data from https://plugins.octoprint.org/notices.json
2025-03-17 23:12:45,981 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/master.zip
2025-03-17 23:12:46,139 - octoprint.plugin.core - INFO - Plugin Filament Sensor Simplified (0.3.3) is disabled.
```

[image]

danieleborgo, 2025-11-28:
Hi, sorry for the delay. I am trying to understand why your Raspberry does not get the data. I need to run few tests

ownconnect, 2025-12-10:
Hi,
I experienced a similar problem on my RPi3 and figured out, that the plugin’s parser seems to be outdated.
At least for my problem :-) My Stacktrace looked a bit different.

This line in the plugin: `config[4][config[4].find("=") + 1:]`
is blowing up on my setup with: `IndexError: list index out of range`
because config is based on: `config = line.split()`
My `raspi-gpio get 0-27` output looks like: `GPIO 0: level=1 func=INPUT`
Splitting that gives:
```
["GPIO", "0:", "level=1", "func=INPUT"]
# indexes: 0      1      2          3
```
There is no config[4] anymore. It seems, that older raspi-gpio versions used to output extra fields (like fsel=0 pull=UP), so the plugin probably assumed at least 5 elements. Newer Raspberry Pi OS trimmed that, so the plugin crashes on every refresh.

Solution which worked for me:
In `octoprint_gpiostatus/__init__.py`
Replace: 
```
def __get_raw_status(self):
    ...
    return [
        [config[1][...], config[2][...], config[4][...]]
        for config in ...
    ]
```
With:
```
def __get_raw_status(self):
        # Call raspi-gpio and parse its current output format, e.g.
        # "GPIO 0: level=1 func=INPUT"
        raw = GPIOStatusPlugin.__execute_command(
            f"raspi-gpio get 0-{self.__max_bcm_pin}"
        )

        parsed = []

        for line in raw.splitlines():
            line = line.strip()
            # Skip empty lines or any non-GPIO noise
            if not line.startswith("GPIO"):
                continue

            parts = line.split()
            # Expect at least: ["GPIO", "0:", "level=1", "func=INPUT"]
            if len(parts) < 4:
                continue

            # Pin number is the "0:" -> strip colon
            pin = parts[1].rstrip(":")

            level = None
            func = None

            # Find tokens like "level=1" and "func=INPUT"
            for token in parts:
                if token.startswith("level="):
                    level = token.split("=", 1)[1]
                elif token.startswith("func="):
                    func = token.split("=", 1)[1]

            # If we didn't find both, skip this line
            if level is None or func is None:
                continue

            parsed.append([pin, level, func])

        return parsed

```
After: `sudo systemctl restart octoprint`
the Plugin shows all GPIO-PINs
:-) :-)
Cheers, Nick

ownconnect, 2026-02-21:
Confirmed: the updated __init__.py from master fixes the issue for me.
GPIO status works again. Not sure about @NilsRo original case though.
Thanks for the fix!